Tactile-electronics/windmill-electronics #1: Charging options

amcewen, 2022-01-28:
It would be useful to have the option to use batteries to smooth out/store the power harvested from wind or solar.

Li-Ion (18650 cells) and/or LiPo seem the most sensible battery options, which would give 3.7-4.1V and are available in various sizes.  The cheap garden solar lights use Ni-MH, but they're only 1.2V so would need to be run in series to get to useful voltages...

Input voltages that allow (the seemingly more common) 6V solar panels, and potentially higher voltages from stepper motors (with a voltage doubler circuit the steppers we've been using were getting up to 19V...) would be good.

Tactile-electronics, 2022-02-01:
Ah ok this makes a lot of sense - are there any charging / balancing issues with  the Li-Ion (18650 cells) type of batteries? 
Can't it get dodgy with LiPo batteries - or has this been blown out of proportion?

amcewen, 2022-02-01:
There don't seem to be balancing issues, at least not that we've encountered using multiple 18650 cells in parallel at Aeternum or for the My Bike's Got LED boards.

You do need to be careful not to overcharge or over discharge LiPo and Li-Ion batteries, but we'd include a protection chip in the circuit (I've got that on the My Bike's Got LED board) to disconnect the battery if that happens.

I've been looking at possible options... got a few Texas Instruments chips and another that we use at Aeternum to add to this—will draw up a table tomorrow so we can compare them.

CN3063.  $0.743 each.  Input voltage 4.4V-6V.  Mentions solar as a potential input.

Actually, the manufacturer of that chip has an assortment of possible options - http://www.consonance-elec.com/product-E-Solar%20Charge%20Management.html - from around $0..40 from LCSC.com.  The CN3722, CN3165, CN3163 and CN3791 seem worth a closer look.

Some other modules I've found on LCSC, probably not as useful as the Consonance modules...

[LP4060S](https://www.lcsc.com/product-detail/Battery-Management-ICs_LOWPOWER-LP4060SB5F_C517260.html) $0.0509.  Input voltage 3.9-6V (with 8V as absolute max rating).  No mention of solar, seemingly intended more for USB charging

[RY4054](https://www.lcsc.com/product-detail/Battery-Management-ICs_RYCHIP-Semiconductor-Inc-RY4054AT5_C910135.html) $0.0746.  4.25-6.5V (max of 10V).  No mention of solar

[SE9019](https://www.lcsc.com/product-detail/Battery-Management-ICs_Seaward-Elec-SE9019_C415995.html) $0.487 4.7-5.5V (max of 6.5V) Mentions MPPT, so presumably is solar-compatible, but the datasheet is in Chinese, so  tricky  to know how to use fully.

LCSC seem to have a bunch of (what I suspect are) TP4056-compatible chips, for around $0.10 each.  Might be worth trying the board I've got here to see how it copes with solar or wind...

amcewen, 2022-02-02:
Finding chips with solar provision seems to be the trickiest part.  

| Chip | Low voltage | High voltage | Price |
|---|---|---|---|
| [AEM10941](https://e-peas.com/product/aem10941/) | 0.05V | 5V | £3.95+vat |
| [BQ24210](https://www.ti.com/lit/ds/symlink/bq24210.pdf?ts=1643621020813&ref_url=https%253A%252F%252Fjlcpcb.com%252F) | 3.5V | 7V (max 18V) | £2.19 |
| [CN3063](http://www.consonance-elec.com/seriesCN3063-E.html#Application_Note%EF%BC%9A) | 4.4V | 6V | $0.743 |
| [CN3163](http://www.consonance-elec.com/seriesCN3163-E.html) | 4.4V | 6V | $0.416 |
| [CN3722](http://www.consonance-elec.com/seriesCN3722-E.html) | 7.5V | 28V | $1.48 |
| [CN3165](http://www.consonance-elec.com/seriesCN3165-E.html) | 4.4V | 6V | $0.474 |
| [CN3791](http://www.consonance-elec.com/seriesCN3791-E.html) | 4.5V | 28V | $0.669 |
| [SE9019](https://www.lcsc.com/product-detail/Battery-Management-ICs_Seaward-Elec-SE9019_C415995.html) | 4.7V | 5.5V | $0.49 |

The first two are probably good if we were just using small solar panels or USB power.  They're not as useful for wind power (I think) and are pretty expensive compared to the other options.

The CN3791 looks like the best choice, although that requires a few more components around it than the others.  The low end of the voltage range would ideally be a bit lower, but the issue then is that you'd need some sort of boost circuit to get the voltage above the battery voltage, so I think that's why something just above 4.2V is so common.

Tactile-electronics, 2022-02-08:
Ok - so does the voltage in need to be above 4.5V? 
So would a boost circuit be before or after the chip? 
this is more complex so trying to get my head around it.
is it possible to boost before the chip if needed and after the chip too to for circuits that need a higher voltage?

amcewen, 2022-02-20:
Okay (for an initial test board at least) we're going to go with the CN3791.  Spinning designing the circuit for that out as its own issue — #2

closed 2022-02-20 by amcewen