Tactile-electronics/windmill-electronics #3: Choose MPPT voltage

amcewen, 2022-03-23:
The voltage for the MPPT (maximum power point tracking) for the solar panel is set with a voltage divider of `R3` and `R4`.  We'll need to choose suitable values for that to set the voltage.

From the [datasheet](http://www.consonance-elec.com/pdf/datasheet/DSE-CN3791.pdf):
> The maximum power point voltage is decided by the following equation:
>     Vmppt = 1.205 x (1 + R3 / R4)

amcewen, 2022-05-18:
I'm still not sure what's best for the MPPT voltage.  Looking at [this solar harvesting board I've used in the past](https://www.tindie.com/products/jaspersikken/solar-harvesting-into-li-ion-battery/) for an example, that's set to 70% of the open circuit voltage of the panel.

Now, we don't know what panels we're likely to use, but I do have a 12V panel here I can try, and 6V panels are also quite common, so as a start I'll get some resistors for 70% of either of those voltages.

Reworking that equation we get that R3/R4 = (0.7*Voc)/1.205 - 1.  So for a 12V panel R3/R4 = 5.97; and for a 6V panel R3/R4 = 2.485.  Mapping very roughly onto more common resistor values, if we get some 330ohm, 1K and 2K, we can see how that fares.

huffeec, 2022-07-20:
On the test boards we made up today R3=2K R4=330 which gives R3/R4 ~ 6

amcewen, 2022-08-26:
Reading [this article on MPPT](https://hackaday.com/2022/08/03/maximum-power-point-tracking-optimizing-solar-panels/) I wonder if the capacitors to smooth the wind output are (a) getting in the way of the chip working out the right solar power to track the maximum power point, and (b) if they'd even be needed for the wind power either.

Could be worth some tests in #12 with the capacitors removed.