Tactile-electronics/windmill-electronics #4: Lay out PCB

amcewen, 2022-03-23:
Following on from #2, we need to work out the physical layout for the circuit.

amcewen, 2022-04-24:
Board laid out, including copper fill zones round the parts of the charging circuit that might get warmer.  Added some mounting holes too.

Just needs some extra labels now:
* To mark inputs / outputs
* Explain jumpers (where possible)
* Add a revision number
* Add a link to this repo (plus https://tactile-electronics.tumblr.com/ and mcqn.com/ibal233)
* Add RoHS, WEEE logos
* Add OSHW logo

amcewen, 2022-05-02:
I've finished (I think) the silkscreen labels.  Any tweaks/suggestions/etc. @Tactile-electronics, before I send off for a handful (half-a-dozen?  A dozen?) to get fabbed for testing?

[image]

[image]

Tactile-electronics, 2022-05-03:
Oh wow ,  brilliant! I’m really excited to try them out. I would imagine
there’s not a huge amount of difference price wise so a dozen?
Is it easier for you to order them and me pay you back or the other way
around?
Thanks
Laura

amcewen, 2022-05-05:
Great.  I've ordered 15 from JLC, and most of the parts — still need to get the bridge rectifiers and resistors (and connectors, but I'm not sure we'd want to populate them by default anyway) — from LCSC (needed to get the charger chips from there, so figured I'd order the other easy stuff at the same time)

About £60 so far (it's all in $) and half of that is postage :grin:  It'll be a couple of weeks before it arrives, I'll let you know when it gets here.

closed 2022-05-05 by amcewen