Tactile-electronics/windmill-electronics #13: The charging LED stays on, regardless of charge level

amcewen, 2023-02-22:
I've been using it as a USB charger for a few months now, and while it works well, it never reports that charging is done.

I've had a 10500mAh Li-Ion battery on charge for five days solid, and it's still showing the "charging" LED.  I don't think the battery was especially discharged (might even have been fully charged) before I put it on to charge, and its sitting at 4.22V now so I think it *is* charged.

Obviously it would be better if the "done" LED on the charging circuit worked :grin:

huffeec, 2023-02-27:
The [data sheet](https://www.laskakit.cz/user/related_files/dse-cn3791-2.pdf) has a formula for the 'Full scale charge current' as a function of  Rcs (R5)

Ich = 120mV / Rcs

Charge termination is determined by: "In constant voltage mode, the charge current decreases gradually. When the charge current decreases to 16% of the full-scale current, the charging is terminated"

Under PCB layout considerations it recommends; 
"(6) Place the charge current sense resistor RCS right next to the inductor output but oriented such that the IC’s CSP and BAT traces going to RCS are not long. The 2 traces need to be routed together as a single pair on the same layer at any given time with smallest trace spacing possible."

So it's possible that a modified trace, or different value for R5 could resolve this.

huffeec, 2023-03-13:
The original boards had Rcs as 0.068 ohm
Giving a charging current of 1760mA

[https://www.digikey.co.uk/en/maker/blogs/charging-lithium-ion-batteries](https://www.digikey.co.uk/en/maker/blogs/charging-lithium-ion-batteries) suggests a charge current of about half the capacity of the battery. So for the 13400mAh packs we get ~6700mA

So (I think) we should be aiming for Rcs of ~0.018 Ohm 

I've made up a board with the closest we currently have to that. Two 0.068 stacked in parallel giving 0.034 ohms
It's charging so I'll see if this triggers the 'Done' light

huffeec, 2023-03-15:
Based on what was discovered in #12 the next step is to setup board C with a bench supply and measure current with one of the INA219 sensors.

huffeec, 2023-04-05:
Running an 8 hour test with board A gives this data for the changing current.
The Bus voltage was a fairly constant 3.11V for the whole period.

First thoughts are generally the values are much lower than I was expecting, and I can't really explain why the current is negative after 7hrs.

[image]

huffeec, 2023-04-17:
Test two with board A makes even less sense. This is a five hour series. 
The starting battery voltage was 3.87V, bus voltage remains constant throughout. 
I am starting to think the chip on this board is no longer working. The next test will be with board D
[image]

amcewen, 2023-08-30:
I've been wondering about this some more today.  No real progress, but...
* I wonder if it's worth trying the [CN3781 chip](http://www.consonance-elec.com/en/73.html) instead.  That requires a tiny change to the circuit, which I think could be tested with a modded version of the existing PCBs.
* Paul downstairs has a logic probe that can test for high impedance as well as voltage.  Might be worth confirming the state of the LED pins to check they match table 1 on page 7 of the CN3791 datasheet
* I wonder if a higher charge voltage (I'm assuming we've been mostly testing with USB voltages of 5V so far) would fix the problem (not a full fix, but to identify a possible issue)
* And/or running a test with the 12V solar panel I've got

Beyond that, maybe a different charger chip would be a solution (or the next step to try)?  Presumably something which can cope with an input voltage of ~12V (to allow 12V solar panels, which are quite common), but it might not need the 28V and up to 4A ratings of the existing chip - maybe the fact we're often at the lower-end of its design range is what's causing some of the trouble?

amcewen, 2023-11-02:
A couple more thoughts (courtesy of Paul H):
* check battery chemistry - is the charger chip designed for LiPo but we've been using Li-Ion batteries?
* check for leakage current on battery - is it pulling a bit still so the charger thinks things are still charging?  Maybe the protection circuit (or in reality, likely two protection circuits) is consuming a bit of power and the charger always thinks it's still charging.

I've bought a few of the CN3781 chips to try sometime too.